averycblack/ECEnabler #4: Not working on OC for HP Laptop

ivnnv, 2021-05-01:
Hi,
First of all thanks for this contribution, if this finally fixes almost any laptop battery, that would be a great accomplishment for the Hackintosh community.

Im trying to use the text on OC `0.6.8` MacOS `10.15.7`.
I have copied the kext into `EFI/OC/Kext`.
Created the entry at `config.plist`'s  `Kernel > Add`

```
<dict>
    <key>Comment</key>
    <string></string>
    <key>MaxKernel</key>
    <string></string>
    <key>PlistPath</key>
    <string>Contents/Info.plist</string>
    <key>Enabled</key>
    <true/>
    <key>MinKernel</key>
    <string></string>
    <key>ExecutablePath</key>
    <string>Contents/MacOS/ECEnabler</string>
    <key>Arch</key>
    <string>Any</string>
    <key>BundlePath</key>
    <string>ECEnabler.kext</string>
</dict>
```
But it seems is not working. I it is not showing on Hackintools loaded kexts and neither the battery indicator shows in the menubar.

Am I missing something? How can I send you more information for debugging the issue?

1Revenger1, 2021-05-01:
This relies on Lilu, and I think I set the minimum version to 1.4.9. If this doesn't work, can you please send a log from OC?

ivnnv, 2021-05-02:
Im on Lilu `1.5.2` and this is the OC log:

[opencore-2021-05-02-124257.txt](https://github.com/1Revenger1/ECEnabler/files/6411437/opencore-2021-05-02-124257.txt)

1Revenger1, 2021-05-02:
Hrmm, do you have the battery device appearing in IOReg? Mind sending the DSDT I guess?

1alessandro1, 2021-05-02:
Here on a HP CE2072NL laptop - [repo here](https://github.com/1alessandro1/HP-Pavilion-CE2072NL-macOS) - it works flawlessly

ivnnv, 2021-05-02:
@1Revenger1, that was it! Once I added `SMCBatteryManager.kext` and added the entry on OC's config. BOOM! Battery works!!!

This is amazing, please provide a donation link or something because as I developer, I really appreciate this engineering work.

This issue can be closed now. Version 1.0 of this plugin already works like a charm 👍🏽

1Revenger1, 2021-05-02:
> This is amazing, please provide a donation link or something because as I developer, I really appreciate this engineering work.

I try to not taking donations out of principle, I like doing this in my free time and I expect nothing out of it. I appreciate the gesture though 👍 

Glad to hear it working though!

closed 2021-05-02 by 1Revenger1