averycblack/ECEnabler #5: Can't be used for Lenovo T570

andreszerocross, 2021-05-07:
I try to use this kext, and remove all ACPI Patch for Battery (16Byte Patch). 
I try to use new DSDT from ACPI Dump. But still battery is not showing.
When i applied Battery Patch for this model in DSDT then Battery is working.

Bootloader : OpenCore 0.7.0
macOS : Catalina 10.15.7

andreszerocross, 2021-05-07:
I close this issue for now,,

It's weird, battery just disapeared in Recovery Menu or in first stage macOS Installation, and then show after installation.

It looks the kext is not working in Recovery Mode (just my opinion)

closed 2021-05-07 by andreszerocross

1Revenger1, 2021-05-07:
Hrmm, it is known that it doesn't work during installation. It should work in recovery though (just tested 10.15.7 recovery). Weird that it didn't work for you.

m0d16l14n1, 2021-05-07:
Some days ago i did reinstall macOS Big Sur, that kext worked in recovery