averycblack/ECEnabler #7: Cycle count doesn't work on Hp probook 430 g3

Human7900, 2021-05-17:
Hi
Thank you so much for this awesome kext.
The cycle count on my hp laptop doesn't work. Because of this I am unable to use the macOS powerchime.
Screenshots attached.

[image]

[image]

1Revenger1, 2021-05-17:
As far as I'm aware, this isn't an issue with ECEnabler. All this does is allow ACPI to read/write fields larger than 8 bits. Check and make sure it isn't something that HP broke in their _BIX method. I've heard that this is broken for a lot of HPs

closed 2021-05-17 by 1Revenger1

Human7900, 2021-05-18:
I do not have any coding skills so I think I will have to live with it . Can you tell me how to check if the _BIX method is broken or not. Thanks once again.