averycblack/ECEnabler #10: Battery manufacturer's name bugged after installing the kext

brkbrlz16, 2021-07-13:
So as the title says, after I installed the kext battery manufacturer name under System Info turned into SÄPˇˇH (used to be SMP). My battery works OOB but I still have many bytes that need patching and I want to use this kext for better stability.

1Revenger1, 2021-07-18:
Hrmm, do you have a copy of your DSDT? Is this with any DSDT patches to your ec regions?
Do you mind trying the version included before? It zeros out the buffer before it's written to when reading from the EC.
[ECEnabler-1.0.1-DEBUG.zip](https://github.com/1Revenger1/ECEnabler/files/6835511/ECEnabler-1.0.1-DEBUG.zip)

brkbrlz16, 2021-07-19:
Only patch I have for EC is the _Q16 to XQ16 rename which is lid open query. It's for YogaSMC's clamshell mode but I've been using that rename even before I added ECEnabler to my config. I also have this SSDT:

`DefinitionBlock ("", "SSDT", 2, "hack", "ECRW", 0x00000000)
{
    External (_SB_.PCI0.LPCB.EC0_, DeviceObj)
    External (_SB_.PCI0.LPCB.EC0_.BAT0, DeviceObj)

    Scope (_SB.PCI0.LPCB.EC0)
    {
        Method (RE1B, 1, NotSerialized)
        {
            OperationRegion (ERAM, EmbeddedControl, Arg0, One)
            Field (ERAM, ByteAcc, NoLock, Preserve)
            {
                BYTE,   8
            }

            Return (BYTE) /* \_SB_.PCI0.LPCB.EC0_.RE1B.BYTE */
        }

        Method (RECB, 2, Serialized)
        {
            Arg1 = ((Arg1 + 0x07) >> 0x03)
            Name (TEMP, Buffer (Arg1){})
            Arg1 += Arg0
            Local0 = Zero
            While ((Arg0 < Arg1))
            {
                TEMP [Local0] = RE1B (Arg0)
                Arg0++
                Local0++
            }

            Return (TEMP) /* \_SB_.PCI0.LPCB.EC0_.RECB.TEMP */
        }

        Method (WE1B, 2, NotSerialized)
        {
            OperationRegion (ERAM, EmbeddedControl, Arg0, One)
            Field (ERAM, ByteAcc, NoLock, Preserve)
            {
                BYTE,   8
            }

            BYTE = Arg1
        }

        Method (WECB, 3, Serialized)
        {
            Arg1 = ((Arg1 + 0x07) >> 0x03)
            Name (TEMP, Buffer (Arg1){})
            TEMP = Arg2
            Arg1 += Arg0
            Local0 = Zero
            While ((Arg0 < Arg1))
            {
                WE1B (Arg0, DerefOf (TEMP [Local0]))
                Arg0++
                Local0++
            }
        }

        Method (NBAT, 0, Serialized)
        {
            If (CondRefOf (BAT0))
            {
                Notify (BAT0, 0x80) // Status Change
            }
        }
    }
}
`

I tried the kext you attached but it's still the same. Here's my original DSDT extracted from Windows using SSDT Time.

[orig-LEN-DSDT.aml.zip](https://github.com/1Revenger1/ECEnabler/files/6842013/orig-LEN-DSDT.aml.zip)

1Revenger1, 2021-07-20:
Can you enable debug logs for the kext and see if you can capture when it tries to read the manufacturer?
Best bet is to use `msgbuf=1048576 -ecedbg` in boot-args and then use `sudo dmesg | grep ECE` in terminal once your booted (do it relatively quickly after booting).

It seems that the manufacturer is 72 bits long, or 9 bytes long. This should be fine, but I'm curious what is sent to the handler in AppleACPIEC

brkbrlz16, 2021-07-21:
[    0.802509]: ECEnabler      init: @ (DBG) ECEnabler bootstrap DBG-MDL-2021-07-17
[    0.967909]: ACPI: ACPI: SSDT 0x000000007CECE000 000346 (v01 LENOVO CB-01    00000001 ACPI 00040000)SSDT 0x000000007CECE000 000346 (v01 LENOVO CB-01    00000001 ACPI 00040000)ACM: Env_SetDataProvider: data provider *set* for var[25].
[    2.336807]: ECEnabler       ECE: @ (DBG) write @ 0x0 of size: 0x8
[    2.337546]: ECEnabler       ECE: @ (DBG) read @ 0x0 of size: 0x8
[    2.338283]: ECEnabler       ECE: @ (DBG) write @ 0x0 of size: 0x8
[    2.349351]: ECEnabler       ECE: @ (DBG) read @ 0x0 of size: 0x8

Here's the output.

1Revenger1, 2021-07-21:
Your EC doesn't have any regions above 8 bits.
```
            OperationRegion (ERAM, EmbeddedControl, Zero, 0xFF)
            Field (ERAM, ByteAcc, Lock, Preserve)
            {
                VCMD,   8, 
                Offset (0x60), 
                SMPR,   8
            }
```

Most of them are for SystemMemory, which does not have a restriction on it's size afaik
If you don't mind testing more though to see why it's erroring out, do you mind giving macOS version and testing the version below? I'm not exactly sure what the error is, this shouldn't be messing with anything except the writes to VCMD, but I don't see anywhere where it's written to in your DSDT.

brkbrlz16, 2021-07-21:
Sure, I can do anything if it's gonna help you with troubleshooting and making the kext even better.

`[    0.663763]: ECEnabler      init: @ (DBG) ECEnabler bootstrap DBG-MDL-2021-07-17
[    0.665145]: ACPI: ACPI: SSDT 0x000000007CECE000 000346 (v01 LENOVO CB-01    00000001 ACPI 00040000)SSDT 0x000000007CECE000 000346 (v01 LENOVO CB-01    00000001 ACPI 00040000)
[    0.986540]: ECEnabler       ECE: @ (DBG) write @ 0x0 of size: 0x8
[    0.986550]: ECEnabler       ECE: @ (DBG) 0x10 0x10
[    0.992055]: ECEnabler       ECE: @ (DBG) read @ 0x0 of size: 0x8
[    0.995837]: ECEnabler       ECE: @ (DBG) write @ 0x0 of size: 0x8
[    0.995845]: ECEnabler       ECE: @ (DBG) 0x1a 0x1a
[    0.997888]: ECEnabler       ECE: @ (DBG) read @ 0x0 of size: 0x8
`

That's the output. I'm running the latest build of Big Sur which is 11.4 20F71.