averycblack/ECEnabler #11: ECEnabler Random ACPI Errors

martinlhzy, 2021-08-14:
I'm getting random errors during boot with ECEnabler related to battery. Sometimes there are more, sometimes there are less, and even times when there's nothing.

With the manual ACPI patching I had no problem.

Even if the errors are present sometimes, the battery indicator working.

martinlhzy, 2021-08-14:
`
2021-08-14 14:50:38.419062+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Error:
2021-08-14 14:50:38.419064+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Error:
2021-08-14 14:50:38.419397+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Method parse/execution failed
2021-08-14 14:50:38.419399+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Method parse/execution failed
2021-08-14 14:50:38.420218+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) [\_SB.PCI0.LPCB.EC0.BTST] (Node ffffff9365dc1240)
2021-08-14 14:50:38.420220+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) [\_SB.PCI0.LPCB.EC0.BTST] (Node ffffff9365dc1240)
2021-08-14 14:50:38.421520+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_BAD_PARAMETER
2021-08-14 14:50:38.421522+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_BAD_PARAMETER
2021-08-14 14:50:38.422001+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
2021-08-14 14:50:38.422003+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
2021-08-14 14:50:38.424112+0200 0x1f0      Default     0x0                  0      0    kernel: DSMOS has arrived
2021-08-14 14:50:38.429131+0200 0x344      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Matched against SMC (started)
2021-08-14 14:50:38.429134+0200 0x344      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Generation from SMC report as 2
2021-08-14 14:50:38.441696+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Error:
2021-08-14 14:50:38.441698+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Error:
2021-08-14 14:50:38.442022+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Method parse/execution failed
2021-08-14 14:50:38.442023+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Method parse/execution failed
2021-08-14 14:50:38.442825+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) [\_SB.BTST] (Node ffffff9365daadb0)
2021-08-14 14:50:38.442826+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) [\_SB.BTST] (Node ffffff9365daadb0)
2021-08-14 14:50:38.443763+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_BAD_PARAMETER
2021-08-14 14:50:38.443764+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_BAD_PARAMETER
2021-08-14 14:50:38.444246+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
2021-08-14 14:50:38.444247+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
2021-08-14 14:50:38.454089+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Error:
2021-08-14 14:50:38.454090+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Error:
2021-08-14 14:50:38.454412+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Method parse/execution failed
2021-08-14 14:50:38.454412+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Method parse/execution failed
2021-08-14 14:50:38.455216+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) [\_SB.BAT0._BST] (Node ffffff9365daa2a0)
2021-08-14 14:50:38.455217+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) [\_SB.BAT0._BST] (Node ffffff9365daa2a0)
2021-08-14 14:50:38.456286+0200 0x1f0      Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_BAD_PARAMETER
2021-08-14 14:50:38.456287+0200 0x74       Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_BAD_PARAMETER
2021-08-14 14:50:38.456768+0200 0x74       Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
2021-08-14 14:50:38.456770+0200 0x74       Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
`

1Revenger1, 2021-08-14:
Can you send the DSDT?

martinlhzy, 2021-08-14:
[DSDT.zip](https://github.com/1Revenger1/ECEnabler/files/6986848/DSDT.zip)

Inside you will find my DSDT and my previous battery patch.

1Revenger1, 2021-08-14:
Thanks, can you give a more complete log?
Something like `sudo dmesg | grep -i ACPI` or `log show --last boot | grep -i ACPI` should work fine.

I see the function where it's returning bad parameter, but not which value it reads which gives that error.

martinlhzy, 2021-08-14:
[log.zip](https://github.com/1Revenger1/ECEnabler/files/6986948/log.zip)