averycblack/ECEnabler #21: Can't compile

ellaosx2007, 2022-06-27:
macOS 12.4
Xcode 13.2.1
Lilu debug with Ventura support.
MacKernelSDK

Im getting 3 errors:
1. Architecture not supported
2. Symantec Issue: (Unknown type name)
3. Error: too many errors emitted, stopping now

1Revenger1, 2022-06-27:
Is Lilu and MacKernelSDK at the root of the project? Should be able to build it using `xcodebuild -jobs 1 -configuration Debug` or `xcodebuild -jobs 1 -configuration Release` in terminal without any other configuration. This is what the CI does, though I think CI runs on Big Sur still.

onixsosx, 2022-06-28:
Compiling in Xcode selecting **Product > Build For > Profiling** gave the te errors i have mentioned.
But if i compile in Xcode selecting **Product > Build** gave me 828 warning but compiled successfully. compiling using terminal using  **xcodebuild -jobs 1 -configuration Release** also gave me the same warning, but compiled successfully.

Are thses warnings can be ignored?

1Revenger1, 2022-06-28:
Ah, never tried `Build For > Profiling`. I'll double check it later. The warnings can be ignored.

onixsosx, 2022-06-28:
BTW @1Revenger1 
can i just use the build from **Actions > Artifacts**? 
I dont know much about the use of Aritifacts, but are these no different from compiling my own?