averycblack/ECEnabler #22: ECEnabler.kext + SMCBatteryManager.kext + VirtualSMC.kext, Battery status and info power in system info gone

oldman20, 2022-07-21:
[image]

Hi, my ACPI EC registers 8 bit
Battery status and info power in system report fine with ACPIBatteryManager.kext 1.90.1 (latest)

But when I tried replace ACPIBatteryManager.kext with ECEnabler.kext + SMCBatteryManager.kext + VirtualSMC.kext, Battery status and info power in system info gone

I checked kexts loaded and missing SMCBattery, don’t know why?

```
% kextstat| grep -i -e Lilu -e ECEnabler -e SMCBattery

Executing: /usr/bin/kmutil showloaded

No variant specified, falling back to release

   48    7 0                  0x2f000    0x2f000    as.vit9696.Lilu (1.6.1) 7FE96C72-5555-3D86-BA9C-36D9177168FB <8 6 5 3 2 1>

   53    0 0                  0x8000     0x8000     com.1Revenger1.ECEnabler (1.0.3) 8C5BDAE0-6618-3F41-B0DC-B9A07648A117 <48 8 6 5 3 2 1>
```

1Revenger1, 2022-07-21:
If your using OpenCore, the order of kexts matter in your Config.plist. SMCBattery needs to be after VirtualSMC, and VirtualSMC + ECEnabler need to be after Lilu.

oldman20, 2022-07-24:
> If your using OpenCore, the order of kexts matter in your Config.plist. SMCBattery needs to be after VirtualSMC, and VirtualSMC + ECEnabler need to be after Lilu.

Thank, so propertree not correctly oder kext automatic? but still so detail like ACPIBatteryManager show detail in IORegistryExplorer
[image]

While SMCBattery:
[image]

dreamwhite, 2022-11-15:
Any update on this?

5T33Z0, 2022-12-03:
> Any update on this?

Read this for clarification: https://github.com/5T33Z0/OC-Little-Translated/tree/main/01_Adding_missing_Devices_and_enabling_Features/AC_Adapter_(SSDT-AC)#acpibatterymanager-vs-smcbatterymanager

oldman20, 2022-12-04:
> > Any update on this?
> 
> Read this for clarification: https://github.com/5T33Z0/OC-Little-Translated/tree/main/01_Adding_missing_Devices_and_enabling_Features/AC_Adapter_(SSDT-AC)#acpibatterymanager-vs-smcbatterymanager

Thank