averycblack/ECEnabler #23: Menu bar not showing battery until power adapter pluged in

kleqing, 2022-07-26:
I hackintosh on my laptop for 2 hours ago. Everything are working perfectly except the battery. At first i think that i should use DSDT. But when in plug the power adapter into my laptop, the battery icon are showing after i pluged the AC in. There's anyway to fixed it? Thanks you! Sorry about my bad english.

1alessandro1, 2022-08-25:
Did you include VirtualSMC and plugins alongside EcEnabler? Share your config.plist or EFI

kleqing, 2022-08-26:
@1alessandro1  I'm using virtualsmc with smcbatterymanager.kext. I don't know why my battery icon is missing until i pluged the power cable in. This is my EFI I've built for 3 days ago but I'm no longer using hackintosh any more. I think that i will reinstall macOS soon again if some one can help me fixed this.
[EFI.zip](https://github.com/1Revenger1/ECEnabler/files/9430013/EFI.zip)

krupama, 2022-08-28:
I have the same problem. Battery indicator after a fresh boot is only appearing after i plug or unplug the powercable. After that works as expected until the next reboot.

kleqing, 2022-08-28:
@krupama I had read some guides and they said that reset EC could solve the problem. But after reset EC (Unpluged AC and battery, then press power button for 30 secs). The problem still there. Have you tried it? I think there's another way to reset it. Did you have any idea?

krupama, 2022-08-28:
@kleqing Thanks for the hint. Performing an EC reset (removed battery, held power button for 30 sec.) did seem to fix the issue for me!

kleqing, 2022-08-29:
@krupama Same here! I tried reset EC for the first time and it work! But after a reboot. The problem still there. When I reset EC again, it didn't work anymore! I think that the problem maybe come from the battery, not the EC.

MinecraftNerrd1337, 2024-04-10:
I got the exact same problem with my Sony Vaio SVE151G11M

kleqing, 2024-04-10:
It was almost 2 years ago and I really forgot it. But yeah, I have a solution
 - If your laptop isn't Sony VAIO, you can try reset EC
 - If your laptop is Sony VAIO, I recommend use DSDT patched. (It isn't complicate but use DSDT is not compatible with OpenCore! use OpenCore No ACPI instead)

MinecraftNerrd1337, 2024-10-11:
I made a working SSDT + hot patches for my Vaio, it works flawlessly now. If anyone has a similar Vaio and the same issue, I can help you solve it