averycblack/ECEnabler #24: SMC Battery manger hanging

Edwardwich, 2022-12-18:
I have problem with VirtualSMC battery kext :
I use SMC Battery kext 
I add Ecenable kext but nothing change 

sometime when laptop connect to ac :
[image]

and charging light when back to red (charging battery) :
[image]
battery person Percentage not change only way back to work unplug and plug agin

ioreg :
[ali’s MacBook Pro.zip](https://github.com/1Revenger1/ECEnabler/files/10254952/ali.s.MacBook.Pro.zip)
apci:
https://github.com/Edwardwich/hp-DA1023nia-macOS/tree/main/Opencore-Debug/SysReport/ACPI
config:
[config.plist.zip](https://github.com/1Revenger1/ECEnabler/files/10254946/config.plist.zip)

1Revenger1, 2023-02-16:
No, I've got no clue why this would happen. All ECEnabler does is read big ACPI fields.

Edwardwich, 2023-02-16:
> No, I've got no clue why this would happen. All ECEnabler does is read big EC fields.

ok TNX

MinecraftNerrd1337, 2024-10-11:
What device is it?

Edwardwich, 2024-10-11:
@1Revenger1 HP DA1023nia its a main laptop I use
https://github.com/Edwardwich/hp-DA1023nia-macOS

MinecraftNerrd1337, 2024-12-05:
As @1Revenger1 said, all ECEnabler does is read big EC fields. Some devices will still require further ACPI patching, which can be a mess because it's completely different for every device.

I will need to do this for the Lenovo Legion I'm currently writing from (15ACH6H) because battery status freezes after wake from sleep. So I'll learn sh*t about this topic and possibly end up being able to help people with it

Edwardwich, 2024-12-06:
> As @1Revenger1 said, all ECEnabler does is read big EC fields. Some devices will still require further ACPI patching, which can be a mess because it's completely different for every device.
>
> I will [...]

ok I need ur help all ssdt and dsdt of my laptop I post it on my GitHub repo :
https://github.com/Edwardwich/hp-DA1023nia-macOS/tree/main/Opencore-Debug/SysReport