averycblack/ECEnabler #30: It doesn't work on Lenovo Legion 5 Ryzen 5 4600H(15arh05).

irfan-vora, 2024-03-08:
I have latest version of this kexts Lilu.kext, VirtualSMC.kext, SMCBatteryManager.kext still not showing the battery but after that I tried this ECEnabler.kext but there is still no sign of battery status. Please help me how to fix this. I am using this kext on ventura. Please help me to fix it.

MinecraftNerrd1337, 2024-10-11:
It works for me on 15ACH6H w/ R7 5800H so that's strange.
Does it start working when you plug charger by any chance?

irfan-vora, 2025-02-02:
Hey @MinecraftNerrd1337,

Sorry for late reply, I fixed that thing while ago I don’t know at the moment what I did but I know I fixed it. So, I am closing this issue here as it got resolved.

closed 2025-02-02 by irfan-vora