averycblack/ECEnabler #32: Battery status not working on Thinkpad T14 G1 AMD

PSzczepanski1996, 2024-08-21:
Hiya!

I got a Thinkpad T14 G1 with AMD Ryzen 7 4750U CPU.
For some reason the battery status is not working even if I use `ECEnabler.kext` and `SMCBatteryManager.kext`.
Anyway I have battery section visible in the settings panel, but there is no any percentage of battery shown.
I also have SSDTTime patches from `https://chefkissinc.github.io/` guide.

[image]

This is my `config.plist` file:
[config.plist.zip](https://github.com/user-attachments/files/16689435/config.plist.zip)

Do you got any idea what I do wrong?

PSzczepanski1996, 2024-08-29:
[image]
[image]
[image]

Log file:

[battery_log.txt](https://github.com/user-attachments/files/16802928/battery_log.txt)

Edwardwich, 2024-10-22:
> Updated yesterday to Sequoia 15.0.1 and issue still persists.

try acpi battery manager kext