averycblack/ECEnabler #34: Not working on ideapad Gaming 3-15ACH6 Laptop BIOS H3CN47WW

Najip, 2025-05-07:
Hello sir. Thanks for your amazing kext.
But unfortunately, This kext isn't working on on ideapad Gaming 3-15ACH6 Laptop BIOS.

I'm using ACPIBatteryManager.kext by Rehabman since Lilu + VirtualSMC + SMCBatteryManager.kext isn't working.
TBH, DSDT of my laptop is very weird. It has single internal battery but named as BAT1 (not BAT0). The placement of the Device is also not inside ``EC```( ```SB.PCI0.LPC0.EC0```) but at ```SB.PCI0.LPC0``` (as ```SB.PCI0.LPC0.BAT1```).
So weird right?

I also attach my DSDT.

[DSDT-Ideapad-Gaming-3-H3CN47WW.dsl.zip](https://github.com/user-attachments/files/20075222/DSDT-Ideapad-Gaming-3-H3CN47WW.dsl.zip)

1Revenger1, 2025-05-07:
Huh, wild. Your EC (or something communicating with the EC) is memory mapped instead of using I/O ports. All the battery function calls lead back to the ECMM operation region which is in system memory (instead of being a EmbeddedControl like I would normally expect).
Anyways, ECEnabler won't help with this due to that difference. You'll likely need to grab logs from the ACPI drivers in macOS to see if any errors are being thrown. `log show --last boot` should grab them if you are looking for a place to start.

closed 2025-05-07 by 1Revenger1

Najip, 2025-05-07:
I guess it also has something to do with those issues regarding of many Lenovo don't have working battery.

Anyway, thanks for your dedication.